nopnop2002/esp-idf-parallel-tft #6: 0xFFFF White color appears to be mint green on ILI9486

Jalxie, 2021-02-03:
# Enviroment 
- Development Kit: ESP32-DevKitC
- Kit version: v4
- Module or chip used: ESP32-WROOM-32D
- IDF version: ESP-IDF v4.1 5ef1b3900 
- Build System: CMake
- Operating System: Windows 10 Pro 19042.746
- Power Supply: USB

# Problem Description
On [3.5inch Arduino Display-UNO](http://www.lcdwiki.com/3.5inch_Arduino_Display-UNO), the color 0xFFFF appears to be mint green. 
[image]
[image]
[image]
[image]
[image]

- The screen uses ILI9486 driver.
- Default Pin configuration

nopnop2002, 2021-02-03:
Is the RGB color bar displayed correctly?

[image]

Jalxie, 2021-02-03:
Yes. 
[image]

nopnop2002, 2021-02-03:
Please tell me this result:

https://github.com/prenticedavid/MCUFRIEND_kbv/tree/master/examples/LCD_ID_readreg

Jalxie, 2021-02-03:
Sorry, could you please explain a little bit?  Run the code on ESP32?

nopnop2002, 2021-02-03:
number.
This is for Arduno R3.

Maybe your TFT is not LI9486.
It becomes clear if you read the register of TFT.

Jalxie, 2021-02-03:
Ok, got it. But don't have an Arduino at the moment.  

I'll see if I can get one.

Thanks.

nopnop2002, 2021-02-03:
Do not use this on the ESP32 as the GPIO on the ESP32 is not 5V tolerant.

closed 2023-02-05 by nopnop2002