nopnop2002/esp-idf-parallel-tft #12: Esp32s2 compilation error.  Because it has only one I2S

lanceingit, 2021-05-16:
Error compiling i2s_lcd.c, when setting target to esp32.
Because esp32s2 has only one I2S, and the register and esp32 have changes, such as the lack of pdm_conf

nopnop2002, 2021-05-17:
Thank you for reporting.

The ESP32S2 compilation error has been fixed.

However, I found that i2s does not work with ESP32S2.
I don't know why.

There is no information about i2s parallel mode of esp32s2.
Please let me know if you know anything.

lanceingit, 2021-05-17:
I've only recently written about screen drivers. I'm not familiar with I2S parallel mode.
But I used the [esp-iot-solution](https://github.com/espressif/esp-iot-solution) repository, and they worked. Maybe you can refer to it.

nopnop2002, 2021-05-18:
Thank you for the information.

I replaced the i2s LCD driver with the latest one.

Works fine with esp32s2.

closed 2021-05-18 by nopnop2002