nopnop2002/esp-idf-parallel-tft #17: 16-bit parallel port support?

modi12jin, 2021-08-18:
# I found in your code that it seems to support 16-bit parallel port. How do I start it?

[image]

# I made some official efforts but failed on 16-bit parallel

https://github.com/espressif/esp-iot-solution/issues/122

nopnop2002, 2021-08-18:
Change here to 16.
https://github.com/nopnop2002/esp-idf-parallel-tft/blob/main/components/tft_library/lcd_com.c#L317

Uncomment here.
https://github.com/nopnop2002/esp-idf-parallel-tft/blob/main/components/tft_library/lcd_com.c#L317

Add define. 
BOARD_LCD_I2S_Dn_PIN.

modi12jin, 2021-08-19:
[image]

@nopnop2002 How to change this code

nopnop2002, 2021-08-19:
```
void gpio_lcd_write_data(int dummy1, unsigned char *data, size_t size) {
#if 8bit_mode
	for (int i=0;i<size;i++) {
		gpio_digital_write(LCD_D0_PIN, data[i] & 0x01);
		gpio_digital_write(LCD_D1_PIN, data[i] & 0x02);
		gpio_digital_write(LCD_D2_PIN, data[i] & 0x04);
		gpio_digital_write(LCD_D3_PIN, data[i] & 0x08);
		gpio_digital_write(LCD_D4_PIN, data[i] & 0x10); 
		gpio_digital_write(LCD_D5_PIN, data[i] & 0x20);
		gpio_digital_write(LCD_D6_PIN, data[i] & 0x40);
		gpio_digital_write(LCD_D7_PIN, data[i] & 0x80);  
		gpio_set_level( LCD_WR_PIN, 0 );
		gpio_set_level( LCD_WR_PIN, 1 );
	}

#else
	for (int i=0;i<size/2;i=i+2) {
		gpio_digital_write(LCD_D0_PIN, data[i] & 0x01);
		gpio_digital_write(LCD_D1_PIN, data[i] & 0x02);
		gpio_digital_write(LCD_D2_PIN, data[i] & 0x04);
		gpio_digital_write(LCD_D3_PIN, data[i] & 0x08);
		gpio_digital_write(LCD_D4_PIN, data[i] & 0x10); 
		gpio_digital_write(LCD_D5_PIN, data[i] & 0x20);
		gpio_digital_write(LCD_D6_PIN, data[i] & 0x40);
		gpio_digital_write(LCD_D7_PIN, data[i] & 0x80);  
		gpio_digital_write(LCD_D8_PIN, data[i+1] & 0x01);
		gpio_digital_write(LCD_D9_PIN, data[i+1] & 0x02);
		gpio_digital_write(LCD_D10_PIN, data[i+1] & 0x04);
		gpio_digital_write(LCD_D11_PIN, data[i+1] & 0x08);
		gpio_digital_write(LCD_D12_PIN, data[i+1] & 0x10); 
		gpio_digital_write(LCD_D13_PIN, data[i+1] & 0x20);
		gpio_digital_write(LCD_D14_PIN, data[i+1] & 0x40);
		gpio_digital_write(LCD_D15_PIN, data[i+1] & 0x80);  
		gpio_set_level( LCD_WR_PIN, 0 );
		gpio_set_level( LCD_WR_PIN, 1 );
	}

#endif

}
```

```
void reg_lcd_write_data(int dummy1, unsigned char *data, size_t size) {
#if 8bit_mode
	for (int i=0;i<size;i++) {
		reg_digital_write(LCD_D0_PIN, data[i] & 0x01);
		reg_digital_write(LCD_D1_PIN, data[i] & 0x02);
		reg_digital_write(LCD_D2_PIN, data[i] & 0x04);
		reg_digital_write(LCD_D3_PIN, data[i] & 0x08);
		reg_digital_write(LCD_D4_PIN, data[i] & 0x10); 
		reg_digital_write(LCD_D5_PIN, data[i] & 0x20);
		reg_digital_write(LCD_D6_PIN, data[i] & 0x40);
		reg_digital_write(LCD_D7_PIN, data[i] & 0x80);	
		gpio_set_level( LCD_WR_PIN, 0 );
		gpio_set_level( LCD_WR_PIN, 1 );
	}
#else
	for (int i=0;i<size/2;i=i+2) {
		reg_digital_write(LCD_D0_PIN, data[i] & 0x01);
		reg_digital_write(LCD_D1_PIN, data[i] & 0x02);
		reg_digital_write(LCD_D2_PIN, data[i] & 0x04);
		reg_digital_write(LCD_D3_PIN, data[i] & 0x08);
		reg_digital_write(LCD_D4_PIN, data[i] & 0x10); 
		reg_digital_write(LCD_D5_PIN, data[i] & 0x20);
		reg_digital_write(LCD_D6_PIN, data[i] & 0x40);
		reg_digital_write(LCD_D7_PIN, data[i] & 0x80);	
		reg_digital_write(LCD_D8_PIN, data[i+1] & 0x01);
		reg_digital_write(LCD_D9_PIN, data[i+1] & 0x02);
		reg_digital_write(LCD_D10_PIN, data[i+1] & 0x04);
		reg_digital_write(LCD_D11_PIN, data[i+1] & 0x08);
		reg_digital_write(LCD_D12_PIN, data[i+1] & 0x10); 
		reg_digital_write(LCD_D13_PIN, data[i+1] & 0x20);
		reg_digital_write(LCD_D14_PIN, data[i+1] & 0x40);
		reg_digital_write(LCD_D15_PIN, data[i+1] & 0x80);	
		gpio_set_level( LCD_WR_PIN, 0 );
		gpio_set_level( LCD_WR_PIN, 1 );
	}

}
```

nopnop2002, 2021-08-19:
```
//#define BOARD_LCD_I2S_BITWIDTH 8
#define BOARD_LCD_I2S_BITWIDTH 16
```

```
void gpio_lcd_write_data(int dummy1, unsigned char *data, size_t size) {
	if (BOARD_LCD_I2S_BITWIDTH == 8) {
		for (int i=0;i<size;i++) {
			gpio_digital_write(LCD_D0_PIN, data[i] & 0x01);
			gpio_digital_write(LCD_D1_PIN, data[i] & 0x02);
			gpio_digital_write(LCD_D2_PIN, data[i] & 0x04);
			gpio_digital_write(LCD_D3_PIN, data[i] & 0x08);
			gpio_digital_write(LCD_D4_PIN, data[i] & 0x10); 
			gpio_digital_write(LCD_D5_PIN, data[i] & 0x20);
			gpio_digital_write(LCD_D6_PIN, data[i] & 0x40);
			gpio_digital_write(LCD_D7_PIN, data[i] & 0x80);  
			gpio_set_level( LCD_WR_PIN, 0 );
			gpio_set_level( LCD_WR_PIN, 1 );
		}
	} else {
		for (int i=0;i<size/2;i=i+2) {
			gpio_digital_write(LCD_D0_PIN, data[i] & 0x01);
			gpio_digital_write(LCD_D1_PIN, data[i] & 0x02);
			gpio_digital_write(LCD_D2_PIN, data[i] & 0x04);
			gpio_digital_write(LCD_D3_PIN, data[i] & 0x08);
			gpio_digital_write(LCD_D4_PIN, data[i] & 0x10); 
			gpio_digital_write(LCD_D5_PIN, data[i] & 0x20);
			gpio_digital_write(LCD_D6_PIN, data[i] & 0x40);
			gpio_digital_write(LCD_D7_PIN, data[i] & 0x80);  
			gpio_digital_write(LCD_D8_PIN, data[i+1] & 0x01);
			gpio_digital_write(LCD_D9_PIN, data[i+1] & 0x02);
			gpio_digital_write(LCD_D10_PIN, data[i+1] & 0x04);
			gpio_digital_write(LCD_D11_PIN, data[i+1] & 0x08);
			gpio_digital_write(LCD_D12_PIN, data[i+1] & 0x10); 
			gpio_digital_write(LCD_D13_PIN, data[i+1] & 0x20);
			gpio_digital_write(LCD_D14_PIN, data[i+1] & 0x40);
			gpio_digital_write(LCD_D15_PIN, data[i+1] & 0x80);  
			gpio_set_level( LCD_WR_PIN, 0 );
			gpio_set_level( LCD_WR_PIN, 1 );
		}
	}

}
```

```
void reg_lcd_write_data(int dummy1, unsigned char *data, size_t size) {
	if (BOARD_LCD_I2S_BITWIDTH == 8) {
		for (int i=0;i<size;i++) {
			reg_digital_write(LCD_D0_PIN, data[i] & 0x01);
			reg_digital_write(LCD_D1_PIN, data[i] & 0x02);
			reg_digital_write(LCD_D2_PIN, data[i] & 0x04);
			reg_digital_write(LCD_D3_PIN, data[i] & 0x08);
			reg_digital_write(LCD_D4_PIN, data[i] & 0x10); 
			reg_digital_write(LCD_D5_PIN, data[i] & 0x20);
			reg_digital_write(LCD_D6_PIN, data[i] & 0x40);
			reg_digital_write(LCD_D7_PIN, data[i] & 0x80);	
			gpio_set_level( LCD_WR_PIN, 0 );
			gpio_set_level( LCD_WR_PIN, 1 );
		}
	} else {
		for (int i=0;i<size/2;i=i+2) {
			reg_digital_write(LCD_D0_PIN, data[i] & 0x01);
			reg_digital_write(LCD_D1_PIN, data[i] & 0x02);
			reg_digital_write(LCD_D2_PIN, data[i] & 0x04);
			reg_digital_write(LCD_D3_PIN, data[i] & 0x08);
			reg_digital_write(LCD_D4_PIN, data[i] & 0x10); 
			reg_digital_write(LCD_D5_PIN, data[i] & 0x20);
			reg_digital_write(LCD_D6_PIN, data[i] & 0x40);
			reg_digital_write(LCD_D7_PIN, data[i] & 0x80);	
			reg_digital_write(LCD_D8_PIN, data[i+1] & 0x01);
			reg_digital_write(LCD_D9_PIN, data[i+1] & 0x02);
			reg_digital_write(LCD_D10_PIN, data[i+1] & 0x04);
			reg_digital_write(LCD_D11_PIN, data[i+1] & 0x08);
			reg_digital_write(LCD_D12_PIN, data[i+1] & 0x10); 
			reg_digital_write(LCD_D13_PIN, data[i+1] & 0x20);
			reg_digital_write(LCD_D14_PIN, data[i+1] & 0x40);
			reg_digital_write(LCD_D15_PIN, data[i+1] & 0x80);	
			gpio_set_level( LCD_WR_PIN, 0 );
			gpio_set_level( LCD_WR_PIN, 1 );
		}
	}
}
```

modi12jin, 2021-08-19:
@nopnop2002 I have tried and failed

> The 16-bit parallel port of the official board may have something worthy of reference

[image]

https://github.com/espressif/esp-dev-kits/tree/master/esp32-s2-hmi-devkit-1

closed 2023-07-13 by modi12jin