nopnop2002/esp-idf-parallel-tft #18: QUESTION: SPI or parallel for new designs?

marcdraco, 2021-09-15:
Not sure how else to post this, so I hope this isn't too off-topic. Let's assume that I'm designing a new board (or two) based on an ESP32 module. I have one of the WROOM devboards - but the new design will likely use an S3 when they drop.

Assuming there are enough free GPIOs left, is parallel preferable over SPI? (I'm only enough to remember the days that we thought parallel was the Dog's Danglers and Serial was consigned to telephones. (Data skew? What's that?! Electrons will never travel that fast and IBM PCs will never penetrate the home!)

So yeah, serial has a place - but I'm curious, is there any particular reason to use SPI over parallel on a new design (one that I can just drop a TFT into?)

My only experience is with the Arduino framework and that largely insulates us from the nitty-gritty but this needs I2S and some serious beef so short of going full SBC with all the uses they come with, I figured the ESP32 is perfect.

nopnop2002, 2021-09-15:
It is case by case.

# Performance comparison using ILI9341(240x320)
SPI used [this](https://github.com/nopnop2002/esp-idf-ili9340).   

|Test|SPI|GPIO parallel|REGISTER I/O parallel|I2S parallel|
|:-:|:-:|:-:|:-:|:-:|
|FillTest|1620|2700|1920|1560|
|ColorBarTest|80|420|160|50|
|ArrowTest|250|460|170|140|
|LineTest|2690|1040|420|1530|
|CircleTest|2400|980|410|1370|
|RoundRectTest|2400|980|400|1390|
|RectAngleTest|5960|2010|940|6720|
|TriangleTest|6550|2120|990|7630|
|DirectionTest|420|520|200|240|
|HorizontalTest|990|720|300|580|
|VerticalTest|990|710|300|580|
|FillRectTest|160|730|300|120|
|ColorTest|240|850|330|190|
|BMPTest|1600|1930|1320|960|
|JPEGTest|2540|2940|2650|2530|
|PNGTest|2830|3210|2940|2810|

marcdraco, 2021-09-15:
Oh wow. Thank you so much! I can see what you mean - it literally is a case of picking the display that best suits the application. 

Fonts really are a bind. I developed a couple of high performance variants of the Adafruit GFX routines for my last project and fonts are truly a cluster ... of pixels. That's one of the reasons I've moved from the ATMega328p from that project to the ESP32 for the current one (two actually). 

Given that I'm sorely tempted to investigate customising the PCB to carry a full-sized parallel Arduino HAT. There's already a Streamdeck-like design with an TFT mounted in place but I haven't managed to get my hands on one yet to see if it will work for these projects which are aimed at schools.