nopnop2002/esp-idf-parallel-tft #19: Add ST7789 support

modi12jin, 2021-09-22:
[image]

[image]

[image]

> I bought the ST7789 parallel port 8-bit screen on Taobao or AliExpress, and it was successfully lit.

https://user-images.githubusercontent.com/40233017/134378077-3bf54008-d855-4e2b-ab9c-f8909773ea83.mp4

[image]

>  This is my attempt on ESP32S2

> My code comes from the official warehouse of Espressif which supports ST7789. I believe yours should also support it. I hope you can transplant it.

[image]

[image]

[image]

https://gith/espressif/esp-iot-solution

modi12jin, 2021-09-22:
> If someone wants to use this screen, they need to pay attention to the screen interface and backlight voltage

[image]

[image]

nopnop2002, 2021-09-22:
I found this on AliExpress.

Can I buy this converted PCB on AliExpress?

[image]

modi12jin, 2021-09-23:
> I drew the screen adapter board by myself, but it did not solve the backlight problem. I used a boost module, and it is better to use a backlight chip because of the constant current.

[image]

[image]

DD1718PA with pins 3-18V turn to positive and negative 5V 6V 9V 12V 15V 24V DC Step Up Boost Converter Power Supply Module

> Screen interface adapter board

[image]

[image]

[image]

[image]

> lceda screen interface adapter board project

[Project_AXT524124转接板_2021-09-23_11-13-02.zip](https://github.com/nopnop2002/esp-idf-parallel-tft/files/7214849/Project_AXT524124._2021-09-23_11-13-02.zip)

> Touch adapter board can be bought ready-made

[image]

nopnop2002, 2021-09-23:
Thank you for information.

modi12jin, 2021-09-23:
# 1_Screen ST7789V (8080 parallel port 8-bit) pin connection relationship with ESP32-S2 pin

| ST7789V pin number |     LCD function    | ESP32-S2 pin |
| :-------------: | :--------------: | :----------: |
|        1        |       BL_K       |     GND      |
|        2        |       BL_A       |     15V      |
|        3        |       GND        |     GND      |
|        4        | IOVCC(1.8V/2.8V) |     3.3V     |
|        5        |    VDD(2.8V)     |     3.3V     |
|        6        |       GND        |     GND      |
|        7        |       /CS        |     GND      |
|        8        |      /RSTB       |     3.3V     |
|        9        |        RS        |    GPIO41    |
|       10        |       WRB        |    GPIO40    |
|       11        |       RDB        |     3.3V     |
|       12        |       DB7        |    GPIO7     |
|       13        |       DB6        |    GPIO6     |
|       14        |       DB5        |    GPIO5     |
|       15        |       DB4        |    GPIO4     |
|       16        |       DB3        |    GPIO3     |
|       17        |       DB2        |    GPIO2     |
|       18        |       DB1        |    GPIO1     |
|       19        |       DB0        |    GPIO0     |
|       20        |        TE        |     Hang in the air     |
|       21        |        NC        |     Hang in the air     |
|       22        |        NC        |     Hang in the air     |
|       23        |        NC        |     Hang in the air     |
|       24        |        NC        |     Hang in the air     |

# 2_Touch FT5336 pin and ESP32-S2 pin connection relationship

| FT5336 pin number |  touch function  | ESP32-S2 pin |
| :------------: | :---------: | :----------: |
|       1        |     GND     |     GND      |
|       2        |  INT(1.8V)  |     Hang in the air      |
|       3        |     GND     |     GND      |
|       4        |  SDA(1.8V)  |    GPIO8     |
|       5        |  SCL(1.8V)  |    GPIO9     |
|       6        | RESET(1.8V) |     3.3V     |
|       7        |  VDD(2.8V)  |     3.3V     |
|       8        |     GND     |     GND      |

nopnop2002, 2021-09-23:
Perhaps ST7789 and ILI9341 have the same control code.

Try with ILI9341.

closed 2022-03-12 by modi12jin