nopnop2002/esp-idf-parallel-tft #20: QUESTION: open-smart ILI9320 not support

chaoshendasixiong, 2021-11-24:
i got a [tft lcd ili9320](https://michiel.vanderwulp.be/domotica/Modules/OpenSmartStore-LCD/)
and Wemos D1 ESP32, i have changed RS CS RST pins from 35,34,36 to 15,33,32
menuconfig select 9320, but is doesn't work, always white.
Is it not suitable for OPEN-SMART Products? should i giveup to use another lcd or any recommend driver type?

nopnop2002, 2021-11-25:
I don’t have open-smart ILI9320 2.8 inch TFT.

I2S parallel may not work for OPEN-SMART Products.

Try changing the interface to GPIO parallel.

chaoshendasixiong, 2021-11-25:
oh, i got it, thank u so very much!!!
i just was confused by the silk screen of back borad, the meaning of DB2,3,5 is LCD_D2,LCD_D3, LCD_D5. 
now it's work normal.

closed 2021-11-25 by chaoshendasixiong