nopnop2002/esp-idf-parallel-tft #22: ESP IDF Component Manager

modi12jin, 2022-03-12:
When can I see this item in the Component Manager registry?

nopnop2002, 2022-11-15:
ESP-IDF is currently in transition from V4.4 to V5.0.

These are very different.

We are working to support both V4.4 and V5.0.

The Component Manager registry is inadequate to support both V4.4 and V5.0.

modi12jin, 2023-05-26:
@nopnop2002  https://github.com/espressif/idf-extra-components/issues/106

 The official jpeg decoding supports SIMD, but it just takes time.  If done this will be the best performing jpeg codec

 V5 version has been released, I believe your driver will be useful for those who use LVGL.  But I hope your code will be similar to the official template.

nopnop2002, 2023-05-26:
I may not have understood your comment correctly.

Here is the current implementation.

For ESP-IDF V4.4
- ESP32:Using ROM Library
- ESP32S2:Not available
- ESP32S3:Using ROM Library
- ESP32H2:Not supported
- ESP32C2:Not supported
- ESP32C3:Using ROM Library

For ESP-IDF V5.x
- ESP32:Using ROM Library(Don't work IDF Component Registry)
- ESP32S2:Using IDF Component Registry
- ESP32S3:Using ROM Library(Don't work IDF Component Registry)
- ESP32H2:Using ROM Library(Don't work IDF Component Registry)
- ESP32C2:IDF Component Registry
- ESP32C3:Using ROM Library(Don't work IDF Component Registry)

modi12jin, 2023-05-27:
Thank you so much for your patience!

closed 2023-05-27 by modi12jin