nopnop2002/esp-idf-parallel-tft #23: Add support for R61529

modi12jin, 2022-03-13:
[image]

```
https://github.com/JadedTuna/lcd-dma-issue
```

[image]

```
https://github.com/espressif/esp-idf/issues/8085
```

[image]

```
https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test
```

[image]

```
https://github.com/moononournation/Arduino_GFX
```

@nopnop2002 Hello, I wonder if you have any plans for R61529 support?

modi12jin, 2022-03-13:
[image]

[Buy on Aliexpress](https://www.aliexpress.com/item/32940608152.html?spm=a2g0o.productlist.0.0.66e773f6vAhLCI&algo_pvid=f9214500-0ba3-4aff-b9a8-9fc9dc51cd2a&algo_expid=f9214500-0ba3-4aff-b9a8-9fc9dc51cd2a-16&btsid=25887317-95e8-4f7a-aabc-1913fdfeaadf&ws_ab_test=searchweb0_0,searchweb201602_6,searchweb201603_53)

modi12jin, 2022-03-13:
[image]

```
https://daumemo.com/ips-capacitive-lcd-on-an-esp32-diy-generator-part-8/
```

According to the seller, the LCD has IPS panel, which has great viewing angles. Also, it has resolution of 480×320 which might be a bit of a problem when used with not so powerful MCU such as ESP32. It has 8080 (MIPI-DBI Type B) interface for sending image data to the LCD. It also should have an SPI connection, but I haven’t tried it, and what is more, according to the LCD driver’s datasheet, it doesn’t support sending image data through the SPI (only commands for setting up the display).

As mentioned in the description it can be interfaced by 8 or 16 data bit lanes and because ESP32 has a few available pins, 8-bit interface was used. Through this interface you can send 16, 18 or 24 bits per pixel which means that you can get up to 16M of colors from this LCD (provided you have enough of the MCU resources).

@nopnop2002 this might help

modi12jin, 2022-03-13:
[image]

[LiangHaoCai-3.5寸IPS-40p-R61529A-资料包.zip](https://github.com/nopnop2002/esp-idf-parallel-tft/files/8238961/LiangHaoCai-3.5.IPS-40p-R61529A-.zip)

@nopnop2002 China Taobao store, the information given by the seller.

nopnop2002, 2022-03-13:
I can't find "TFT Shield" that using R61529.

modi12jin, 2022-03-13:
@nopnop2002 This screen can be purchased at Buy on Aliexpress

https://es.aliexpress.com/item/32940608152.html?gatewayAdapt=glo2esp&spm=a2g0o.productlist.0.0.66e773f6vAhLCI&algo_pvid=f9214500-0ba3-4aff-b9a8-9fc9dc51cd2a&algo_expid=f9214500-0ba3-4aff-b9a8-9fc9dc51cd2a-16&btsid=25887317-95e8-4f7a-aabc-1913fdfeaadf&ws_ab_test=searchweb0_0,searchweb201602_6,searchweb201603_53

[image]

[image]

[LiangHaoCai-3.5寸IPS-40p-R61529A.pdf](https://github.com/nopnop2002/esp-idf-parallel-tft/files/8239354/LiangHaoCai-3.5.IPS-40p-R61529A.pdf)

modi12jin, 2022-03-13:
https://github.com/daumemo/IPS_LCD_R61529_FT6236_Arduino_eSPI_Test/issues/1

[image]

@nopnop2002 Compatibility may exist

modi12jin, 2022-03-13:
@nopnop2002 My friend tested it and there is compatibility. Thank you

closed 2022-03-13 by modi12jin