nopnop2002/esp-idf-parallel-tft #24: Support for ESP32S3

modi12jin, 2022-03-13:
@nopnop2002 Hi, are there plans for ESP32S3 support?

nopnop2002, 2022-03-13:
Yes,
I will support When i get it.

But it's still expensive.

NIDA11120204, 2022-03-30:
hi, I can send you free samples~

nopnop2002, 2022-03-30:
Currently, ESP-IDF is in the process of migrating to Ver5.

S3 support and Ver5 support should be considered separately.

Still, ESP-IDF Ver5 is very buggy.

Therefore, I will support ESP32-S3 with Ver4.4 in the near future.

NIDA11120204, 2022-03-30:
> Yes, I will support when i get it.
> 
> But it's still expensive.

Could you support for Ai-Thinker ESP-S3-12K and ESP-S3-32S? I can send you the samples

nopnop2002, 2022-03-30:
Ai-Thinker ESP-S3-12K will arrive in the near future.

Support for Ver5 If not included, welcome your sample.

NIDA11120204, 2022-03-30:
> 

may I have your contact?

NIDA11120204, 2022-03-30:
> Ai-Thinker ESP-S3-12K will arrive in the near future.
> 
> Support for Ver5 If not included, welcome your sample.

may I have your contact?

nopnop2002, 2022-03-30:
Unfortunately, I don't publish my email address.

NIDA11120204, 2022-03-30:
> Unfortunately, I don't publish my email address.
> 
> You can publish your sample using your repository

Could you give me your email right now and delete in 2 minutes?

nopnop2002, 2022-03-30:
hau26280@tree.odn.ne.jp

nopnop2002, 2022-03-30:
I give you my email address and delete in 2 minutes

NIDA11120204, 2022-03-30:
> I give you my email address and delete in 2 minutes

you can delete now

nopnop2002, 2022-11-12:
Added support for ESP32S3.

modi12jin, 2022-11-13:
@nopnop2002 Since there is some friction between Espressif and Espressif, your board may become a legacy, they are no longer in production

closed 2022-11-13 by modi12jin

modi12jin, 2022-11-13:
@nopnop2002 They are no longer producing esp32s3, and there is friction with Espressif (maybe because of proxying other MCUs with WiFi)

modi12jin, 2022-11-13:
@nopnop2002 They are no longer making the esp32s3, so your board may become legacy.

modi12jin, 2022-11-13:
@nopnop2002 ESP-S3-12K does not need too much introduction. It should be noted that both flash and psram are QSPI, and both are 8MB.