nopnop2002/esp-idf-parallel-tft #25: Touch input for st7793. open smart 16pin

davidarroyo1234, 2022-04-07:
First of all I wanted to thank you for your work.

I have a question.

I have a open smart 3.5 tft display with this setup.

(Screen pins-> ESP32 pins)
D0 -> 12
D1 -> 13
D2 -> 26
D3 -> 25 
D4 -> 17
D5 -> 19
D6 -> 27
D7 -> 14
RD -> 2
WR -> 4
RS -> 15
CS -> 33
RESET -> 32

Which are the correct pins on each parameter for this to work? Especially the parameter adc_yp and adc_xm
[image]

nopnop2002, 2022-04-07:
You can change any GPIO using menuconfig.

This is default.

# Wireing  

|TFT||ESP32|ESP32S2/S3||
|:-:|:-:|:-:|:-:|:-:|
|LDC_RST|--|GPIO32|GPIO45|*1|
|LDC_CS|--|GPIO33|GPIO42|*1|
|LDC_RS|--|GPIO15|GPIO41|*1|
|LDC_WR|--|GPIO4|GPIO40|*1|
|LDC_RD|--|GPIO2|GPIO39|*1|
|LDC_D0|--|GPIO12|GPIO1|*1 *2|
|LDC_D1|--|GPIO13|GPIO2|*1 *2|
|LDC_D2|--|GPIO26|GPIO3|*1 *2|
|LDC_D3|--|GPIO25|GPIO4|*1 *2|
|LDC_D4|--|GPIO17|GPIO5|*1 *2|
|LDC_D5|--|GPIO16|GPIO6|*1 *2|
|LDC_D6|--|GPIO27|GPIO7|*1 *2|
|LDC_D7|--|GPIO14|GPIO8|*1 *2|
|5V|--|5V|5V|*3|
|3.3V|--|3.3V|3.3V|*3|
|GND|--|GND|GND||

davidarroyo1234, 2022-04-08:
Oh so from the same lcd pin i have to connect to esp32 pins? 1 for the input and 1 for the output?

nopnop2002, 2022-04-08:
Yes
You're right.

from the same lcd pin you have to connect to 2 esp32 pins

```
(X-) ----+---- Gpio for Digital Output
         +---- Gpio for Analog Input

(Y+) ----+---- Gpio for Digital Output
         +---- Gpio for Analog Input
```

davidarroyo1234, 2022-04-08:
Okay, thank you so much. 

I will try this in the afternoon.

nopnop2002, 2022-04-08:
These LCDs are designed for the At Mega328 (Arduino UNO).

With AtMega328 (Arduno UNO), one GPIO can be used by switching between digital output and analog input at runtime.

davidarroyo1234, 2022-04-08:
Thank you :), now it's working.

Im closing the issue.

closed 2022-04-08 by davidarroyo1234