nopnop2002/esp-idf-parallel-tft #26: How to connect ESP-32

joseph-vedadi, 2022-05-12:
Thank you for working on such an important project, 
I am very new to displays and ESP-32, Could you please help me to connect my display and run a simple test 
[image]

[image]
and my board :

[image]

Thank you so much

nopnop2002, 2022-05-12:
# Wireing  

|TFT||ESP32|ESP32S2/S3||
|:-:|:-:|:-:|:-:|:-:|
|LDC_RST|--|GPIO32|GPIO45|*1|
|LDC_CS|--|GPIO33|GPIO42|*1|
|LDC_RS|--|GPIO15|GPIO41|*1|
|LDC_WR|--|GPIO4|GPIO40|*1|
|LDC_RD|--|GPIO2|GPIO39|*1|
|LDC_D0|--|GPIO12|GPIO1|*1 *2|
|LDC_D1|--|GPIO13|GPIO2|*1 *2|
|LDC_D2|--|GPIO26|GPIO3|*1 *2|
|LDC_D3|--|GPIO25|GPIO4|*1 *2|
|LDC_D4|--|GPIO17|GPIO5|*1 *2|
|LDC_D5|--|GPIO16|GPIO6|*1 *2|
|LDC_D6|--|GPIO27|GPIO7|*1 *2|
|LDC_D7|--|GPIO14|GPIO8|*1 *2|
|5V|--|5V|5V|*3|
|3.3V|--|3.3V|3.3V|*3|
|GND|--|GND|GND||

(*1)   
You can change any GPIO using menuconfig.   

[image]

(*2)   
When using REGISTER I/O parallel, GPIO from D0 to D7 is 1 to 31.(GPIO0 is boot mode pin)   

(*3)   
With a regulator (mostly AMS1117) is on the back, the supply voltage is 5V.   
With a regulator is NOT on the back, the supply voltage is 3.3V.

joseph-vedadi, 2022-05-12:
Since I don’t have any other external devices attached, I would be happy with the default one. 
Could you kindly share some example for text and some animations ( like fire or wind ) 

really appreciate it

nopnop2002, 2022-05-12:
There is no animations.
There is example for text.

# Wireing  

|TFT||ESP32|ESP32S2/S3||
|:-:|:-:|:-:|:-:|:-:|
|LDC_RST|--|GPIO32|GPIO45|*1|
|LDC_CS|--|GPIO33|GPIO42|*1|
|LDC_RS|--|GPIO15|GPIO41|*1|
|LDC_WR|--|GPIO4|GPIO40|*1|
|LDC_RD|--|GPIO2|GPIO39|*1|
|LDC_D0|--|GPIO12|GPIO1|*1 *2|
|LDC_D1|--|GPIO13|GPIO2|*1 *2|
|LDC_D2|--|GPIO26|GPIO3|*1 *2|
|LDC_D3|--|GPIO25|GPIO4|*1 *2|
|LDC_D4|--|GPIO17|GPIO5|*1 *2|
|LDC_D5|--|GPIO16|GPIO6|*1 *2|
|LDC_D6|--|GPIO27|GPIO7|*1 *2|
|LDC_D7|--|GPIO14|GPIO8|*1 *2|
|5V|--|5V|5V|*3|
|3.3V|--|3.3V|3.3V|*3|
|GND|--|GND|GND||

(*1)   
You can change any GPIO using menuconfig.   

[image]

(*2)   
When using REGISTER I/O parallel, GPIO from D0 to D7 is 1 to 31.(GPIO0 is boot mode pin)   

(*3)   
With a regulator (mostly AMS1117) is on the back, the supply voltage is 5V.   
With a regulator is NOT on the back, the supply voltage is 3.3V.

joseph-vedadi, 2022-05-12:
Thank you so much .

closed 2022-05-12 by nopnop2002