nopnop2002/esp-idf-parallel-tft #34: Touch not working

Me-Phew, 2024-12-28:
I am using the default wiring pinout with a ILI9486 display. The screen is working correctly but for some reason no touch is ever detected. I have verified all pin numbers and connections. Also touch is actually working correctly without any wiring or pin number modifications using a different library made for Arduino framework.

Relevant part of `sdkconfig`
```
# TFT Configuration
CONFIG_ILI9486=y
CONFIG_INTERFACE_GPIO=y
CONFIG_WIDTH=320
CONFIG_HEIGHT=480
CONFIG_OFFSETX=0
CONFIG_OFFSETY=0
CONFIG_ENABLE_TOUCH=y
CONFIG_ADC_CHANNEL_YP=6
CONFIG_ADC_CHANNEL_XM=7
CONFIG_XP_GPIO_D6=y
CONFIG_XM_GPIO_RS=y
CONFIG_YP_GPIO_WR=y
CONFIG_YM_GPIO_D7=y
# end of TFT Configuration
```

Program startup log:
```
I (594) LCD_COM: LCD_CS_PIN=33
I (594) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (604) LCD_COM: LCD_RS_PIN=15
I (614) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (614) LCD_COM: LCD_WR_PIN=4
I (624) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (634) LCD_COM: LCD_RD_PIN=2
I (634) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (644) LCD_COM: LCD_D0_PIN=12
I (644) LCD_COM: LCD_D1_PIN=13
I (654) LCD_COM: LCD_D2_PIN=26
I (654) LCD_COM: LCD_D3_PIN=25
I (654) LCD_COM: LCD_D4_PIN=17
I (664) LCD_COM: LCD_D5_PIN=16
I (664) LCD_COM: LCD_D6_PIN=27
I (674) LCD_COM: LCD_D7_PIN=14
I (674) LCD_COM: INTERFACE is GPIO
I (674) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (684) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (694) gpio: GPIO[26]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (704) gpio: GPIO[25]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (714) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (724) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (734) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (744) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (754) LCD_COM: LCD_RESET_PIN=32
I (754) gpio: GPIO[32]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (764) main_task: Returned from app_main()
I (964) ILI9486: Your TFT is ILI9486.Same as ST7796
I (964) ILI9486: Screen width:320
I (964) ILI9486: Screen height:480
I (1264) LCD_COM: _adc_yp=0 _adc_xm=1073448112
I (1264) LCD_COM: _gpio_xp=27 _gpio_xm=15
I (1264) TouchGetCalibration: Opening Non-Volatile Storage (NVS) handle...
I (1264) TouchGetCalibration: Opening Non-Volatile Storage (NVS) handle done
I (1274) TouchGetCalibration: nvs_get_i16(calibration)=4354
I (1284) TouchCalibration: TouchGetCalibration=0
I (547014) TouchCalibration: counter=0 _xp=1 _yp=1 xp=-32768 yp=32767
I (547034) TouchCalibration: counter=1 _xp=1 _yp=2 xp=1 yp=1
```

Could `_adc_xm=1073448112` have sth to do with it? It doesn't really look correct.
The counter values shown at the end weren't actually caused by touch but rather by moving the cables which also seems strange.

nopnop2002, 2024-12-31:
Please initialize the NVS and perform calibration again.
```
idf.py erase_flash
```

Me-Phew, 2025-01-09:
Thanks for your answer.

I have managed to resolve most of the issue. Turns out I wasn't using the correct pins for `ADC1`.

That said I've run into some issues trying to fix this.
Namely, when trying to use the correct pins for `ADC1` as stated in this table from the `README`.

> |TFT|ADC1|ESP32|ESP32S2/S3|
> |:-:|:-:|:-:|:-:|
> |LCD-WR (Y+)|Channel#6|GPIO34|GPIO07|
> |LCD-RS (X-)|Channel#7|GPIO35|GPIO08|

The problem was that, at least for my devboard (a generic esp32 with 38 GPIO pins) these pins are input only which makes them impossible to use for `LCD_WR` / `LCD_RS`.

Because of that I ended up using the following modification of the default pinout:

|TFT|ESP32|ADC1|
|:-:|:-:|:-:|
|LDC_WR|GPIO32|Channel <span>#</span>4|
|LDC_RS|GPIO33|Channel <span>#</span>5|
|LDC_RST|GPIO18|-|
|LDC_CS|GPIO19|-|

With that the touch is now finally working.

The screen is also working - for the most part at least. On the first touch calibration screen only the green dots are visible after touch, remaining tests are fully rendered but there is noticeable flickering. My guess is this is happening because I'm using 3v from the esp to power the display and with the adc working it can't supply enough current to the fairly big (3.5") screen. Do you think my assumption is correct or is there something I am still doing wrong?

Anyway, thanks for the nice library. I've also submitted a PR https://github.com/nopnop2002/esp-idf-parallel-tft/pull/35 that you may review if you wish.

nopnop2002, 2025-01-09:
>The problem was that, at least for my devboard (a generic esp32 with 38 GPIO pins) these pins are input only which makes them impossible to use for LCD_WR / LCD_RS.

All ESP32, GPIO pins 34-39 as "input-only".
These pins don't have many uses, so I intentionally made them the default pins for analog input.
The TFT's (X-) and (Y+) require two wires as shown below.

```
LCD-WR(Y+) ----+---- Gpio04(for Digital Output)
               +---- Gpio34(for Analog Input)

LCD-RS(Y+) ----+---- Gpio15(for Digital Output)
               +---- Gpio35(for Analog Input)
```

Will it work with just one wire if We use GPIO32 and 33?
```
LCD-WR(Y+) ----+---- Gpio32(for Digital Output)
               +---- Gpio32(for Analog Input)

LCD-RS(Y+) ----+---- Gpio33(for Digital Output)
               +---- Gpio33(for Analog Input)
```

>My guess is this is happening because I'm using 3v from the esp to power the display and with the adc working it can't supply enough current to the fairly big (3.5") screen. Do you think my assumption is correct or is there something I am still doing wrong?

You are right.
ESP32 development board cannot supply too much current.
It is more stable when supplied from an external power source.
[image]

Me-Phew, 2025-01-10:
>I thought it was not possible to use a single GPIO for both digital output and analog input at the same time.
 Will it work with just one wire if We use GPIO32 and 33?
 
 Is this what you mean by that?

[image]

This is my current working configuration:

[image]

I'm asking this just for clarity because I thought it is impossible to even configure the pins in this way.

[image]

As I understand it now the GPIO numbers are used for digital i/o and are configured completely independently from the ADC channels which are for analog input, right?

nopnop2002, 2025-01-10:
>Is this what you mean by that?

Yes. That right.

>This is my current working configuration:

This is a new discovery for me.

>As I understand it now the GPIO numbers are used for digital i/o and are configured completely independently from the ADC channels which are for analog input, right?

My understanding was that one GPIO could not be used for digital output and analog input at the same time.
This could be wrong, so I'll check again.

nopnop2002, 2025-01-10:
>Namely, when trying to use the correct pins for `ADC1` as stated in this table from the `README`.

 |TFT|ADC1|ESP32|ESP32S2/S3|
 |:-:|:-:|:-:|:-:|
 |LCD-WR (Y+)|Channel#6|GPIO34|GPIO07|
 |LCD-RS (X-)|Channel#7|GPIO35|GPIO08|

What problems will this cause?

You can change this using menuconfig without changing the default value.

[image]

nopnop2002, 2025-02-13:
```
LCD-WR(Y+) ----+---- Gpio32(for Digital Output)
               +---- Gpio32(for Analog Input)

LCD-RS(X-) ----+---- Gpio33(for Digital Output)
               +---- Gpio33(for Analog Input)
```

It turns out that this configuration only works with ESP32.

For ESP32S2/S3, the following configuration is used, but this does not work.
```
LCD-WR(Y+) ----+---- Gpio05(for Digital Output)
               +---- Gpio05(for Analog Input)

LCD-RS(X-) ----+---- Gpio06(for Digital Output)
               +---- Gpio06(for Analog Input)
```